brianfeucht/esphome-balboa-spa #8: Heating state on Climate may be incorrect

brianfeucht, 2025-02-02:
It doesn't seem like my climate state matches what I see with my tub.  

Document the sensor states with what conditions I see with the tub.  Update the climate implementation with those findings.

reicharb, 2025-03-03:
I found this problem also...the climate looks like its heating all the time.

brianfeucht, 2025-03-07:
The heat state sensor seems to line up with energy usage.  This is likely the sensor we will want to use here

[image]

[image]

brianfeucht, 2025-03-07:
Found the issue - https://github.com/brianfeucht/esphome-balboa-spa/commit/523c61cb913387f659627f0983e812c8c5e9880b

brianfeucht, 2025-03-07:
Testing ^ commit on my spa.
Still seeing some weird behavior likely due to https://github.com/brianfeucht/esphome-balboa-spa/issues/11
Increasing the value buffer size seems to be helping make this more stable.

Assuming this continues similar behaviour, I'll merge this to main today

closed 2025-03-07 by brianfeucht