brianfeucht/esphome-balboa-spa #14: Not able to compile

ttrnkocy, 2025-03-03:
I am interested in this project and i would like to use it with ESP32. When trying to compile in ESPHome, i got compiling error, attache log. Could you please help?
Thank you.

[logs_virivka_compile.txt](https://github.com/user-attachments/files/19058162/logs_virivka_compile.txt)

brianfeucht, 2025-03-07:
@ttrnkocy can you please include the YAML you are using?  Please be sure to replace any sensitive strings like passwords or keys

brianfeucht, 2025-03-07:
Seems unlikely this will be the fix, but give it a try and let me know
https://github.com/brianfeucht/esphome-balboa-spa/commit/3a67625e423fbb50a6b88f8a2ea4ecdcc2548712

ttrnkocy, 2025-03-10:
I asked AI where the problem could be. It gave ma an answer, that i should do following:
      default:
            //ESP_LOGD("Spa/Sensors/UnknownSensorType", String((uint8_t)sensor_type, 0).c_str());
            ESP_LOGD("Spa/Sensors/UnknownSensorType", **std::to_string**((uint8_t)sensor_type).c_str());
            // Unknown enum value. Ignore
        return;
Now the compile process passed. Yet i have no access to the whirlpool now, as it is located at my father's house. I will test later.
Thank you.

brianfeucht, 2025-03-11:
Feel free to reopen this issue once you can get me the YAML and ESPHome version you are using.

I'll not be able to provide a fix until I'm able to reproduce this locally.  The AI generated code here doesn't really help me understand how your environment is different than [what I'm doing](https://github.com/brianfeucht/esphome-balboa-spa/blob/main/test_balboa_spa_component.yaml) on my local instance and in our [tests](https://github.com/brianfeucht/esphome-balboa-spa/actions/runs/13725542675/job/38390998229)

lukavia, 2025-03-23:
I have the same issue: 
if I change the file sensors.cpp line 42 from:
```ESP_LOGD("Spa/Sensors/UnknownSensorType", String((uint8_t)sensor_type, 0).c_str());```
to
```ESP_LOGD("Spa/Sensors/UnknownSensorType", "%", String((uint8_t)sensor_type, 0).c_str());```
It then doesn't throws an error there, but there are more like this. 
Current esphome version: 2025.3.1
But it was the same with: 2024.12.2 

Here is my yaml: 
```substitutions:
  devicename: swimspa
  friendlyname: Swim Spa

esphome:
  name: ${devicename}
  friendly_name: ${friendlyname}

esp8266:
  board: esp01_1m

external_components:
  - source:
     type: git
     url: https://github.com/brianfeucht/esphome-balboa-spa
     ref: main

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "~censored~"

ota:
  platform: esphome
  password: "~censored~"

button:
  - platform: safe_mode
    name: "SwimSpa Restart (Safe Mode)"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.20.24
    gateway: 192.168.20.254
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendlyname} Fallback Hotspot"
    password: "~censored~"

captive_portal:

#packages:
#  common: !include .common.yaml

uart:
  id: spa_uart_bus
  tx_pin: GPIO1
  rx_pin: GPIO3
  data_bits: 8
  parity: NONE
  stop_bits: 1
  baud_rate: 115200
  rx_buffer_size: 128

balboa_spa:
  id: spa

switch:
  - platform: balboa_spa
    balboa_spa_id: spa
    jet1:
      name: Jet1
    jet2:
      name: Jet2
    light:
      name: Lights

climate:
  - platform: balboa_spa
    balboa_spa_id: spa
    name: Thermostat
    visual:
      min_temperature: 62 °F    # min: 17
      max_temperature: 105 °F    # max: 30
      temperature_step: 1.0 °F  # min: 0.5

sensor:
  - platform: balboa_spa
    balboa_spa_id: spa
    blower:
      name: Blower
    highrange:
      name: High Range
    circulation:
      name: Circulation
    restmode:
      name: Rest Mode
    heatstate:
      name: Heat State
```