brianfeucht/esphome-balboa-spa #15: Jets have 3 modes - off/low/high - add optional support for all

brianfeucht, 2025-03-07:
Our implementation currently only supports on/off for the jets

Jets actually have 3 states vs the two currently supported.

Add support for the high mode.  It would be great if we can find a way to support this without removing the simple on/off functionality we have now

brianfeucht, 2025-07-07:
I was able to put together some code for this but the cycling the jet toggles doesn't feel great.  I'm going to sit on this for a bit before I merge.  Would love to get feedback on this if you are willing to try it out, especially if this is a feature you would like.

https://github.com/brianfeucht/esphome-balboa-spa/pull/48

Legrand5382, 2025-07-18:
Maybe put the 3 states in a select box ?

brianfeucht, 2025-07-19:
@Legrand5382 thanks for taking a look at this.  Admitting I had Claude.ai vibe code this for me.  I didn't even try to run it because I really didn't love the "keep hitting the button" approach to toggle through the settings.  Basically how it works on the control of the spa.  Seems pretty flaky and error prone to me.

I asked copilot to try and find a different approach and it came up with the same solution.  So maybe this is the only way to do this.  Looking at the serial protocol there isn't a way to send a desired state.  

I'd love to hear how you are planning on using this.  It might make it better to have a desired use case when adding this feature.  I personally have no reason to support this, so it would help me have some context.

robinostlund, 2025-09-26:
Oh this would be so nice to have, when it would be available i could throw out the balboa wifi module which always loose network access 😄 Awesome work with this component, highly appreciated!

eric1090, 2025-10-04:
> On the spa, to choose the speed on pump 1, we press one time, the green led go on, then off and pump is on low, press a seconde time, and now the green led stay on, and the pump is at the second speed, look like this is the way it work. Maybe you can get the current state (0 = off, 1 = low and 2 = high). Automatically update the fan state, and if it go from 1 to 2, then switch one time, from 0 to 2, switch 2 time, from 1 to off, switch to 2, then cycle back to off. By getting the current state of pump 1, you can also validate, if it didn't work and is in the wrong state, then maybe make it compensate ? I don't know if it would work, this is just a suggestion. Thank you for you work !

If i had to competence to integrate that, I would, but still no reply, it's the only thing missing for it to be perfect for me