brianfeucht/esphome-balboa-spa #18: Cannot compile when trying to use esp32

lukavia, 2025-03-23:
Same issue as #14 (I've left a comment there, but cannot reopen the issue so I'm posting a new with reference to the old: 

When trying to compile I get: 
```
INFO ESPHome 2025.3.1
INFO Reading configuration ./swimspa-ext.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing swimspa (board: esp01_1m; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
|-- CircularBuffer @ 1.4.0+sha.b314927
Compiling .pioenvs/swimspa/src/esphome/components/balboa_spa/balboaspa.cpp.o
Compiling .pioenvs/swimspa/src/esphome/components/balboa_spa/sensor/sensors.cpp.o
Archiving .pioenvs/swimspa/libbbc/libsodium.a
Indexing .pioenvs/swimspa/libbbc/libsodium.a
In file included from src/esphome/core/log.h:10,
                 from src/esphome/components/balboa_spa/sensor/sensors.cpp:1:
src/esphome/components/balboa_spa/sensor/sensors.cpp: In member function 'void esphome::balboa_spa::BalboaSpaSensors::update(esphome::balboa_spa::SpaState*)':
src/esphome/components/balboa_spa/sensor/sensors.cpp:42:92: error: initializer fails to determine size of '__pstr__'
   42 |             ESP_LOGD("Spa/Sensors/UnknownSensorType", String((uint8_t)sensor_type, 0).c_str());
/home/lukav/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/WString.h:39:76: note: in definition of macro 'FPSTR'
   39 | #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
      |                                                                            ^~~~~~~~~~~~
...
```
I'm attaching the full log.

[logs_swimspa-ext_run.txt](https://github.com/user-attachments/files/19411596/logs_swimspa-ext_run.txt)

 
Reference from #14 
> I have the same issue: 
> if I change the file sensors.cpp line 42 from:
> ```ESP_LOGD("Spa/Sensors/UnknownSensorType", String((uint8_t)sensor_type, 0).c_str());```
> to
> ```ESP_LOGD("Spa/Sensors/UnknownSensorType", "%", String((uint8_t)sensor_type, 0).c_str());```
> It then doesn't throws an error there, but there are more like this. 
> Current esphome version: 2025.3.1
> But it was the same with: 2024.12.2 
> 
> Here is my yaml: 
> ```substitutions:
>   devicename: swimspa
>   friendlyname: Swim Spa
> 
> esphome:
>   name: ${devicename}
>   friendly_name: ${friendlyname}
> 
> esp8266:
>   board: esp01_1m
> 
> external_components:
>   - source:
>      type: git
>      url: https://github.com/brianfeucht/esphome-balboa-spa
>      ref: main
> 
> # Enable logging
> logger:
>   baud_rate: 0
> 
> # Enable Home Assistant API
> api:
>   encryption:
>     key: "~censored~"
> 
> ota:
>   platform: esphome
>   password: "~censored~"
> 
> button:
>   - platform: safe_mode
>     name: "SwimSpa Restart (Safe Mode)"
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   manual_ip:
>     static_ip: 192.168.20.24
>     gateway: 192.168.20.254
>     subnet: 255.255.255.0
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "${friendlyname} Fallback Hotspot"
>     password: "~censored~"
> 
> captive_portal:
> 
> #packages:
> #  common: !include .common.yaml
> 
> uart:
>   id: spa_uart_bus
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   data_bits: 8
>   parity: NONE
>   stop_bits: 1
>   baud_rate: 115200
>   rx_buffer_size: 128
> 
> balboa_spa:
>   id: spa
> 
> switch:
>   - platform: balboa_spa
>     balboa_spa_id: spa
>     jet1:
>       name: Jet1
>     jet2:
>       name: Jet2
>     light:
>       name: Lights
> 
> climate:
>   - platform: balboa_spa
>     balboa_spa_id: spa
>     name: Thermostat
>     visual:
>       min_temperature: 62 °F    # min: 17
>       max_temperature: 105 °F    # max: 30
>       temperature_step: 1.0 °F  # min: 0.5
> 
> sensor:
>   - platform: balboa_spa
>     balboa_spa_id: spa
>     blower:
>       name: Blower
>     highrange:
>       name: High Range
>     circulation:
>       name: Circulation
>     restmode:
>       name: Rest Mode
>     heatstate:
>       name: Heat State
> ```
>  

 _Originally posted by @lukavia in [#14](https://github.com/brianfeucht/esphome-balboa-spa/issues/14#issuecomment-2746484747)_

brianfeucht, 2025-03-24:
Thanks for posting this and the PR.

I'm trying to reproduce this failure locally and on Github Actions so that I can make sure we keep this fixed long term.

I put together a 8266 test build and it is coming out clean
https://github.com/brianfeucht/esphome-balboa-spa/commit/ca4e36434a2ca228b7dcb8b7af04097b7fa6870b

Just trying to understand why your `ESP_LOGD` is different than what I and the github action have

brianfeucht, 2025-03-24:
I have it failing locally finally.  The cause is:

```
# Enable logging
logger:
  baud_rate: 0
```

lukavia, 2025-03-24:
OMG! I've tried all sorts of things, but I never thought that setting could have such an effect. 
Great job debugging it!
I hope I was of some help nevertheless.

brianfeucht, 2025-03-25:
These contributions were great help!  I'm going to tweak your PR to go all in the with the string formatting.  I'm going to get rid of all the string constructors and replace them with the idf logging lib format.

I am able to reproduce now, so I'll be confident I fix this issue without breaking existing users.

Might be a few days before I get this completed.

brianfeucht, 2025-04-30:
Resolved by https://github.com/brianfeucht/esphome-balboa-spa/pull/24

closed 2025-04-30 by brianfeucht