brianfeucht/esphome-balboa-spa #22: Got fatal error: CircularBuffer.hpp: No such file or directory

rj2s, 2025-04-06:
Hi,
Finally I found this ESPHome Balboa intagration, thks for that.
Just cut-paste your YAML code, on comile it for my ESP32-WROOM-32 and receive this message:

```
   2 | #include <CircularBuffer.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esphome-web-8d1cb0/src/esphome/components/balboa_spa/spa_state.cpp.o] Error 1
In file included from src/esphome/components/balboa_spa/balboaspa.h:8,
                 from src/esphome.h:15,
                 from src/esphome/components/balboa_spa/balboaspa.cpp:2:
src/esphome/components/balboa_spa/spa_state.h:2:10: fatal error: CircularBuffer.hpp: No such file or directory

**************************************************************************
* Looking for CircularBuffer.hpp dependency? Check our library registry!
*
* CLI  > platformio lib search "header:CircularBuffer.hpp"
* Web  > https://registry.platformio.org/search?q=header:CircularBuffer.hpp
*
**************************************************************************
```

see full log in attachement

[logs_esphome-web-8d1cb0_run.txt](https://github.com/user-attachments/files/19623088/logs_esphome-web-8d1cb0_run.txt)

Can you help me solving this ?

Regards

brianfeucht, 2025-04-07:
Can you share your YAML and the commands you are using to compile?  It looks like your build commands are not correctly pulling in external components

That lib gets imported via this include
https://github.com/brianfeucht/esphome-balboa-spa/blob/main/components/balboa_spa/__init__.py#L23

brianfeucht, 2025-04-08:
There isn't a build of circular buffer for idf.

Switch to arduino

```
  framework: 
    type: arduino
```

closed 2025-04-08 by brianfeucht

rj2s, 2025-04-08:
indeed solved by changing type to arduino....
Tx for your support !