brianfeucht/esphome-balboa-spa #51: Support for Secondary Light ("Light 2") in BP Series Devices with Single Jet

dhWasabi, 2025-07-28:
Some BP series devices feature a single jet but include two light channels: a main light and a secondary light labeled as "Light 2". Currently, the secondary light is not supported or exposed in the integration.

I’ve conducted preliminary investigation and confirmed that command "0x12" successfully controls this secondary light.

Enhancement Request:
- Add support for "Light 2" in "switch" with a single jet but dual-light configuration.
- Ensure 0x12 is correctly mapped for handling secondary light control.

This would improve compatibility and allow full use of lighting features on affected BP series models.

[image]

brianfeucht, 2025-08-05:
@dhWasabi I have an implementation completed at https://github.com/brianfeucht/esphome-balboa-spa/pull/54

I do not have a spa with a light2 so I have no way to test.  A big unknown is how the state of light two is encoded into the byte stream.  I'm not seeing any documentation on the [git repo](https://github.com/ccutrer/balboa_worldwide_app/blob/main/doc/protocol.md) with reverse engineering.  Just a note the author only has a single light.  

Feel free to give this a try.  I'm most interested in testing the following:
- Toggling the light switch via the esphome works
- Toggling the light via the spa control updates in esphome.  

If toggling the light via the spa control doesn't update the value in esphome, that tells me we are looking at the wrong bytes.  I'll have to have you do some more debugging if that is the case.

brianfeucht, 2025-08-05:
Some possible other values to try for light2

```
    // Check for second light state - try different options:
    // Option 1: Different bit in same byte
    spa_component_state = bitRead(input_queue[19], 2);
    // Option 2: Different value pattern
    // spa_component_state = input_queue[19] == 0x01;
    // Option 3: Next byte
    // spa_component_state = bitRead(input_queue[20], 0);
    // Option 4: Unused byte 17
    // spa_component_state = bitRead(input_queue[17], 0);
    
    if (spa_component_state != spaState.light2) {
        ESP_LOGD(TAG, "Spa/light2/state: %.0f (byte19=0x%02X)", spa_component_state, input_queue[19]);
        spaState.light2 = spa_component_state;
    }
```

dhWasabi, 2025-08-07:
AH sorry for overlap, just have seen that you have already a draft pull request. let me check that one

brianfeucht, 2025-08-13:
@dhWasabi I pulled in the binary reader code from your `light2` implementation.  `light` was the same on both, so I'm not entirely sure why you are seeing two different behaviors.  

Can you turn on debug level logging and send me the values you are seeing with:
- light1 off & light2 off 
- light1 on & light2 off
- light1 off & light2 on
- light1 on & light2 on

I'm particular interested in the log entries which look like 
```
Spa/light/state: 
Spa/light2/state: 
```

closed 2025-08-19 by brianfeucht