brianfeucht/esphome-balboa-spa #56: Filter cycle *question not issue*

BigBobbas, 2025-08-14:
Hi , 
I am currently looking at solutions to control my Spa via ESPHome. Before I start I want to make sure that I can control the aspects that I need control of.
Are there any plans in the pipeline to add control of the filter cycles. TBH the only 2 things I would like to control (remotley or via automations) are Heat and filter time and duration. Controlling jets is a nice to have but not required personally as they are easily controlled when required from inside the tub.

Many Thanks for your work and I hope to get this up and running soon.

Cheers 
Rob

brianfeucht, 2025-08-14:
Heat (on/off & set value) is controllable via the thermostat interface.  

It is [possible](https://github.com/ccutrer/balboa_worldwide_app/blob/main/doc/protocol.md#filter-cycles-response) to control the filter timers, but I haven't implemented any of that yet.

brianfeucht, 2025-08-14:
Seeing if copilot can get us most of the way there
https://github.com/brianfeucht/esphome-balboa-spa/issues/57