brianfeucht/esphome-balboa-spa #71: Current and target temperature in climate is not showing decimals

robinostlund, 2025-09-27:
Hi,
What i can see the climate current and target temperature doesnt show decimals when setting steps to 0.5c is this expected or is is that the component converts the value to int instead of floats?

Thanks
/ R

robinostlund, 2025-09-27:
Seems to be the same with setting the temperature, 38 works but not 38.5 😊

brianfeucht, 2025-10-06:
@robinostlund I think I pulled in most the changes you made in https://github.com/brianfeucht/esphome-balboa-spa/pull/72

Big change is I moved the client_id override to the yaml.  So you'll want to update your device config to get that.  
```
balboa_spa:
  id: spa
  spa_temp_scale: C
  esphome_temp_scale: C
  uart_id: spa_uart_bus
  client_id: 43
```

Let me know if this resolves the issue for you

closed 2025-10-06 by brianfeucht

robinostlund, 2025-10-07:
Nice, i will try it out when i get some time over ⭐